NabuCasa/esp-web-flasher #46: UART_DATE_REG_ADDR seems wrong for some chips in const.ts

Jason2866, 2021-08-04:
ESP8266: UART_DATE_REG_ADDR = 0x60000078
ESP32: UART_DATE_REG_ADDR = 0x60000080
ESP32C3: UART_DATE_REG_ADDR = 0x6000007c

Source latest dev version of esptool.py

conradopoole, 2021-08-05:
@Jason2866 

UART_DATE_REG_ADDR values are correct for ESP8266 and ESP32. The value that you shared for ESP32 is actually for the ESP32-S3, not for the original ESP32.

For the C3 you are correct, that DATE_REG_ADDRESS should be 0x6000007c, there is a function for retrieving the ADDRESS in const.ts given the chip family, and it is not ready for the C3 family, so it returns -1.

However right now the esp-flasher never uses this register in any operation since it is only used in "writeRegister" when delayAfterUs is greater than 0, and no function currently makes use of this parameter. So while the value is incorrect, it cannot be the reason why flashing to C3 is not working for you.

Can you share more information on your test? Log information to see what chip family is being detected, etc....I tested this with my C3 board when I submitted the PR and it worked, I'll try and re-test again though.

Jason2866, 2021-08-05:
@conradopoole  Thx, for looking into!
I get this with my C3 with latest development esptool.py
```
d:\>esptool.py chip_id
esptool.py v3.2-dev
Serial port COM5
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:a4:4f:c0
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: 7c:df:a1:a4:4f:c0
Hard resetting via RTS pin...
```

Jason2866, 2021-08-05:
```c:\Python37\Scripts>espefuse.exe --port COM5 summary
Connecting....
Detecting chip type... ESP32-C3
espefuse.py v3.1
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_RTC_RAM_BOOT (BLOCK0)                Disables boot from RTC RAM                         = False R/W (0b0)
DIS_ICACHE (BLOCK0)                      Disables ICache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)             Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)              Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                         Disables the TWAI Controller hardware              = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                 Set this bit to vdd spi pin function as gpio       = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                Enable btlc gpio                                   = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                  Set this bit to enable power glitch function       = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)             Sample delay configuration of power glitch         = 0 R/W (0b00)
DIS_LEGACY_SPI_BOOT (BLOCK0)             Disables Legacy SPI boot mode                      = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)              Selects the default UART for printing boot msg     = UART0 R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)              Sets the default UART boot message output mode     = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)               Force ROM code to send a resume command during SPI = False R/W (0b0)
                                          bootduring SPI boot                   
BLOCK_USR_DATA (BLOCK3)                  User data                              
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                          Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                          Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Flash Config fuses:
FLASH_TPUW (BLOCK0)                      Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                          unit is (ms/2). When the value is 15, delay is 7.
                                         5 ms                                   
FLASH_ECC_MODE (BLOCK0)                  Set this bit to set flsah ecc mode.    
   = flash ecc 16to18 byte mode R/W (0b0)
FLASH_TYPE (BLOCK0)                      Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                 Flash page size                                    = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                    Enable ECC for flash boot                          = False R/W (0b0)

Identity fuses:
SECURE_VERSION (BLOCK0)                  Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                         ure)                                   
MAC (BLOCK1)                             Factory MAC Address                    
   = 7c:df:a1:a4:4f:c0: (OK) R/W
WAFER_VERSION (BLOCK1)                   WAFER version                                      = 3 R/W (0b011)
PKG_VERSION (BLOCK1)                     Package version                                    = 8 R/W (0x8)
BLOCK1_VERSION (BLOCK1)                  BLOCK1 efuse version                               = 4 R/W (0b100)
OPTIONAL_UNIQUE_ID (BLOCK2)(0 errors):   Optional unique 128-bit ID             
   = 67 7f 07 db 6e ab 06 cf 23 d3 2e 58 98 3f b7 91 R/W
BLOCK2_VERSION (BLOCK2)                  Version of BLOCK2                                  = 2 R/W (0b010)

Jtag Config fuses:
JTAG_SEL_ENABLE (BLOCK0)                 Set this bit to enable selection between usb_to_jt = False R/W (0b0)
                                         ag and pad_to_jtag through strapping gpio10 when b
                                         oth reg_dis_usb_jtag and reg_dis_pad_jtag are equa
                                         l to 0.                                
SOFT_DIS_JTAG (BLOCK0)                   Software disables JTAG. When software disabled, JT = 0 R/W (0b00)
                                         AG can be activated temporarily by HMAC peripheral
DIS_PAD_JTAG (BLOCK0)                    Permanently disable JTAG access via pads. USB JTAG = False R/W (0b0)
                                          is controlled separately.             

Security fuses:
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)     Disables flash encryption when in download boot mo = False R/W (0b0)
                                         des                                    
SPI_BOOT_CRYPT_CNT (BLOCK0)              Enables encryption and decryption, when an SPI boo = Disable R/W (0b000)
                                         t mode is set. Enabled when 1 or 3 bits are set,di
                                         sabled otherwise                       
SECURE_BOOT_KEY_REVOKE0 (BLOCK0)         If set, revokes use of secure boot key digest 0    = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE1 (BLOCK0)         If set, revokes use of secure boot key digest 1    = False R/W (0b0)
[... cut ...]
DIS_USB (BLOCK0)                         Disables the USB OTG hardware                      = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                  Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)           Disables use of USB in UART download boot mode     = False R/W (0b0)

Vdd_Spi Config fuses:
PIN_POWER_SELECTION (BLOCK0)             GPIO33-GPIO37 power supply selection in ROM code   = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                   Selects RTC WDT timeout threshold at startup       = False R/W (0b0)
```

Jason2866, 2021-08-29:
Fixed. Thx

closed 2021-08-29 by Jason2866