NabuCasa/esp-web-flasher #83: [ESP32] Change to dio/40MHz in bootlader image might cause endless boot loop

wladwnt, 2021-12-31:
ESP Web Flasher alsways change the fourth byte in the bootloader image to 0x20 (dio, 40 MHz). 
It is probably fine for Arduino-ESP32 V1.0.x/esp-idf 3.3 based firmware.
But apparently Arduino-ESP32 V2.0.2/esp-idf 4.4 based firmware do not accept this change and can not be executed.
If it is already based on bootloader that has 0x20, then nothing is changed by ESP Web Flasher during flashing process and it works fine. But if it has in original bootloader for example 0x2f (dio, 80 MHz), then ESP Web Flasher changes it during flashing process to 0x20 and the frimware cannot be started, it stays in endless Watchdog reset loop.

[https://github.com/esphome/esp-web-tools/issues/151](https://github.com/esphome/esp-web-tools/issues/151)

Jason2866, 2022-01-02:
In does not make sense either. In manifest.json all the needed files are provided.
By patching from esp-web-flasher the careful done setup is borked.

TD-er, 2022-01-02:
> At least it needs to be documented.

Or made into a setting.

balloob, 2022-01-03:
Can we just remove this code ?

TD-er, 2022-01-03:
I think so.
Only thing I am not sure about is what to do with ESP32 boards that may have a 26 MHz crystal.

N.B. ESPtool (used when flashing from the IDE) also does seem to set the detected flash size.

ESP32 code for decoding both nibbles:
```c++
uint32_t EspClass::magicFlashChipSize(uint8_t byte)
{
    switch(byte & 0x0F) {
    case 0x0: // 8 MBit (1MB)
        return (1_MB);
    case 0x1: // 16 MBit (2MB)
        return (2_MB);
    case 0x2: // 32 MBit (4MB)
        return (4_MB);
    case 0x3: // 64 MBit (8MB)
        return (8_MB);
    case 0x4: // 128 MBit (16MB)
        return (16_MB);
    default: // fail?
        return 0;
    }
}

uint32_t EspClass::magicFlashChipSpeed(uint8_t byte)
{
    switch(byte & 0x0F) {
    case 0x0: // 40 MHz
        return (40_MHz);
    case 0x1: // 26 MHz
        return (26_MHz);
    case 0x2: // 20 MHz
        return (20_MHz);
    case 0xf: // 80 MHz
        return (80_MHz);
    default: // fail?
        return 0;
    }
}
```

TD-er, 2022-01-03:
Just found this: [Notice of upcoming change: ESP-IDF will assume 40MHz XTAL frequency by default](https://esp32.com/viewtopic.php?t=2800)
Apparently the default crystal frequency was set to 40 MHz in 2017.
So I guess it is unlikely there are ESP32 boards out there using 26 MHz crystal.
If so, then a user may want to compile a version specifically for that board (or set it to auto)

This makes it seem a good choice to just remove the code to change some nibbles when flashing.

closed 2022-01-03 by balloob