NabuCasa/esp-web-flasher #104: Sync with upstream

balloob, 2022-03-09:
- [ ] https://github.com/adafruit/Adafruit_WebSerial_ESPTool/pull/22

balloob, 2022-03-09:
Now that upstream has caught up on compression, it might be worth to explore the set of changes necessary to take current upstream and export it as an NPM package, instead of expanding our version with their latest changes, which have diverged some bits. The goal of this fork is to do the least amount of work to get their version NPM installable.

balloob, 2022-03-25:
I've taken a look at if it's feasible to bundle and package up upstream. From a quick glance there are a few minor issues. Documented it with patch in [this branch](https://github.com/NabuCasa/esp-web-flasher/tree/export-patch/export).

Last time Adafruit was not interested in making it a package, however I think that some of the discovered issues could maybe be fixed upstreamed.

makermelissa, 2022-04-05:
Upstream is now archived, but some of the improvements from there could be added here. :)

closed 2022-04-05 by balloob